wnew/fencing_scoring_box #7: Implement and test short circuit feature for foil

wnew, 2015-09-08:
Currently provision has been made for the short circuit feature for foil. This just needs to be implemented and tested. I am not aware of any such feature for epee. Mmm.... I could be wrong.

wnew, 2015-09-09:
This has been implemented. It only checks every once a second as this is not a highly time critical function and we dont need the loop checking for a short circuit 20000 times a second.

nova0205, 2024-06-14:
hi, did you fix the short circuit bug? where can i found that code?

acoaco2, 2024-06-14:
hi!

actually I don't remember! I try to recover that project and maybe let you
know.

nova0205, 2024-06-14:
oh if you could found it will be so usefull to me thanksss

nova0205, 2024-06-28:
i thougth that you already solved that bug, i will work on that