wnew/fencing_scoring_box #9: Sabre Whipover

wnew, 2015-09-22:
This still needs to be implemented. This is not easy and is a low priority.

wnew, 2017-03-14:
This requirement by the FIE seems to have been removed in favour of a stiffer blade.

Solexis2001, 2017-03-14:
Not removed : it has been added. In 2000, blade were required to be stiffer in order to reduce the whipover possibilities.
But in Book 3 (material rules of FIE), Annex B, Chap 3a7 you can read : "The apparatus must not signal a hit made by the blade whipping over to hit the opponent while in contact with his blade or guard."
And the procedure to avoid signalling a whipping over is described in Chap 3b5 and 6 :
"If the contact between the blade and the opponent’s target takes place ‘through the blade’, the apparatus:
— will register the hit between 0 and 4 ms (+ 1 ms);
— will prevent the hit being registered between 4 and 15 ms (+ 5 ms), on condition that the contact between the two blades is not interrupted more than a maximum of 10 times in the interval.
6. Should there be a hit made by the whipping over of the blade which has not been signalled, whatever method has been used to prevent the signalling, after 15 ms (± 5 ms) from the contact of the blade with the valid target (the time for the registering of the whip) and unless there has been another hit, the apparatus should allow the normal registering of any subsequent hits."
Hope this helps.
Quentin

Solexis2001, 2017-03-14:
test

wnew, 2017-03-14:
So I guess this is back open.  :( Ill have to do some reading. Thanks @Solexis2001