wnew/fencing_scoring_box #10: Sabre Implementation

wnew, 2015-12-02:
Sabre is totally untested, it has come to my attention that it is more complex then just foil without the off-target. A hit needs to be able to register while in contact with the opponents blade. This makes it quite tricky to implement. Hopefully Ill have some time to look at this in the new year. I also need to get hold of sabre equipment to test it.

Solexis2001, 2016-02-27:
I did a sabre box that perfectly worked on my bench but that completely fails when you connect it to long wires (i used digital pins). The principle was very easy : one weapon connected at a time in the loop so when one weapon is high, the opponent's lamé is checked for a high. The contact with the opponent's blade is not a problem as only one weapon is connected at a time and, during that time, only the opponent's lamé is checked for a high.
You can easily simulate a sabre on your test bench with a foil body wire : they're the same for sabre. And the sabre only shorts the circuit between the two wires that goes to the "hand plug". A simple nude copper wrapped around the pins of the hand plug will simulate a sabre :-)
I'm really impatient to test your sabre box. Did you have time to look at it ?

wnew, 2016-02-27:
Hi Solexis, thank you for you interest. I do currently have some time. So will look at it. I am expecting this latest implementation to work as I spent a lot of time on designing it. But Ill test with body wires this weekend and let you know the results.

Please do implement the box and let me know your thoughts. I have also put together an instructable for it. (Assuming that isnt where you came across this repository)

duvalj18, 2016-04-06:
Hey I was wondering, how did it go? I am a saber fencer hoping to build a test strip for the school I go to.

wnew, 2016-04-07:
I have tested sabre with body wires on the bench. But dont have the weapons to test it on a piste. Would you be able to build and test it for me? If there are any issues we can work together to solve them.

duvalj18, 2016-04-07:
Yes sounds good. I will most likely build it over the weekend. I will let you know.

scooper260, 2016-06-21:
Did you get anywhere with the sabre? I only fence sabre, but if you use the arduino to poll the pins in a similar way to how you would a keypad, surely that be a work around?

wnew, 2016-06-22:
So Sabre is implemented but only bench tested. ie. I havent taken it onto the piste yet. I have read about the B and C wires possibly being tied together which could cause some problems. Please try it out and I can help you debug any issues that you find.

mcleadou, 2017-01-01:
I am just about to embark on this journey, sadly with less to offer in technical advice but lots as a fencer. My recent thoughts concern your difficulties with sabre, specifically the whip-over issue. As my early days in fencing precedes the availability of electric sabre scoring of any sort, you have already fashioned avast improvement compared to our situation in those days. 

It has occurred to me that the initial contact between the blades in regular sparring would seldom be continuous. Given the flexibility of modern sabre blades their initial contact must be momentary, and that the reaction of the one blade against another - in an equal and opposite reaction type of thing - must deform both blades both blades for a significant amount of time. So even if there is no light during the instant of contact, it seems to me that many "hits" do register in spite of the intention of the whip-over rules.

Being at the outset of my investment in this project, and a little deficient in electronics knowledge in general, I am not sure whether the clash of blades produces an identifiable event which could be used to initiate a routine which would sample the current directions for a specified amount of time and negate a hit made within that period. That span of time would have to conclude very quickly, as a legitimate remise could conceivably be executed, in a very short window and would need to be counted. 

I have no idea if these ideas present an avenue towards resolving concerns you have with the sabre routine, but I'll be very interested in the countinuing devlopments and discussion in this area.

cdodier, 2017-03-14:
I just found your project and I too am interested in the devellopment of the sabre part. I have all the sabre gear available including 2 different FIE approved scoring equipment :  leon paul and favero. I don't have good knowledge in programing exept for scripting. I just ordered a arduino starting kit to start the same project before finding yours.... So i am very happy to have found it. I will build it and test the saber part. If i understand correctly the whip over issue. You dont want to register a hit when the blade bend over the opponent's weapon, is that it ?   Because current scoring system do register this as a hit for sabre.

Did you implement the double touch lockout (acceptable delay) with all three weapons ? I dont have the exact delay for all weapons in mind but I can provide you with saber  :  120ms +- 10ms

Lockout example :  touch is registered by fencer A, 
fencer B as 120 ms to score. after that only the hit from fence A registers.

mcleadou, 2017-03-14:
Actually - Current FIE timings for Sabre lockout is set at 170ms. This was changed, I believe in the Fall of 2016, and all manufacturers of scoring boxes have upgraded chips available. Changing on the arduino project simply involves changing the lockout timing in the code.

mcleadou, 2017-03-14:
Are you in fact M. C Dodier, from Quebec, Canada, also a sabreur. I am Doug McLean from Ontario. If so we haven't met, but I see your name atop the rankings list..... Nice! 

I have completed the project once, and almost finished my second box. My experience was that the circuits were very good for foil and epee, but that Sabre doesn't register hits accurately at all. If you can code the Sabre part, I'd be very interested in hearing about this.

wnew, 2017-03-14:
@cdodier Thanks for the interest in the project. I would be grateful for any help getting the sabre implementation up and running. Please contact me if I can be of assistance. I would like to get this project to a point where it is fully FIE compliant but dont have a lot of time to put into it at the moment. I would be happy to do the coding if you could provide the specifications and do some testing for me. Lets chat.

Great news on the whipover. I was unaware of this.

Yes, the double hit is implemented. You can see the timings (in micro seconds) in the allweaponsbox.ino file:
const long lockout [] = {300000,  45000, 120000};  // the lockout time between hits
const long depress [] = { 14000,   2000,   1000};     // the minimum amount of time the tip needs to be depressed

You can see https://github.com/wnew/fencing_scoring_box/blob/master/hardware/circuit for the circuit diagram. Please let me know if you have any questions.

wnew, 2017-03-14:
@mcleadou Thanks for your input. Can you provide a link to the documentation with the new timings in it? Then I can update the code. Thanks

wnew, 2017-03-14:
I need to experiment with capacitors between the ground and the long lines to the fencers. This would remove any unexpected spikes on the lines.

Solexis2001, 2017-03-14:
About the whipover part : in 2000, blades were required to be stiffer in order to reduce the whipover possibilities.
But in Book 3 (material rules of FIE 2016), Annex B, Chap 3a7 you can read : "The apparatus must not signal a hit made by the blade whipping over to hit the opponent while in contact with his blade or guard."
And the procedure to avoid signalling a whipping over is described in Chap 3b5 and 6 :
"If the contact between the blade and the opponent’s target takes place ‘through the blade’, the apparatus:
— will register the hit between 0 and 4 ms (+ 1 ms);
— will prevent the hit being registered between 4 and 15 ms (+ 5 ms), on condition that the contact between the two blades is not interrupted more than a maximum of 10 times in the interval.
6. Should there be a hit made by the whipping over of the blade which has not been signalled, whatever method has been used to prevent the signalling, after 15 ms (± 5 ms) from the contact of the blade with the valid target (the time for the registering of the whip) and unless there has been another hit, the apparatus should allow the normal registering of any subsequent hits."
But anyway, on a very practical point of view, most sabrers ignore that thei apparatus discriminates the whipover hits, because they happen very rarely. I think this is not a priority for this box.
And about the problems due to inconsistant hit registering in sabre, I can say that I built your box, that I met this problem, that I tried to add a few capacitors, but my knowledge of electronics theory is so poor that the most visible results of my experiments with capacitors is that when I touch the opponent'ts lame, the point is registered... on HIS side :-) Obviously, to much capacity :-) 
Hope this helps.
Quentin

wnew, 2017-03-14:
@Solexis2001 Thanks for the info. It seems at thought there is enough interest in Sabre for me to put some time into it. Ill try and do this in the next month and keep everyone posted. If anyone does some work on Sabre in the meantime. Please let me know.

mcleadou, 2017-03-14:
For the start of the 2016-2017 season the FIE made two adjustments for Sabre. One change which has since been abandoned was to alter the en garden position of fencers from front foot 4 metres apart to back foot 4 metres apart. This setup was casually known as the Russian Box of Death. As mentioned this change has been revoked. 

The other change was an adjustment of the lockout timing from 120ms p/m (10ms) to 170 ms (p/m). This change has not been revoked and clubs are all over the map when it comes to replacing the chips in commercially sold machines. 

Both changes are described in an FIE bulletin here:

http://static.fie.org/uploads/9/49267-NEW%20RULES%20FOR%20THE%20SABER.pdf

I hope the issues with Sabre can be sorted. Sadly I'm not really much help with the electronics issues.

cdodier, 2017-03-15:
I just received my Arduino kit. It may take a while for me to build it I am kind of a self learner.  My interest in electronic and programing will drive this project. 

I will test the whip over with both of my scoring system and let you know what I find. I’m pretty sure they register as a hit even with FIE approved scoring device.