wnew/fencing_scoring_box #11: My implementation of your box

senjabl, 2018-02-25:
Hey man, thanks.

We made first test of our fencing scoring box, inspired by your project. There is another arduino controlling timer, remote and display. I also 3D printed reels.

When I catch some time, i will publish my files.

You can see it in action on
https://youtu.be/ZYt0c06OIwA

[image]

[image]

luczynski, 2018-03-02:
OK-very good- please send me Your PCB plate -score- ? how much your model-
rules cable ?-meybe we start cooperate  made in Poland and Banja Luka  ?
see You in Poland or Banja luka ?
02.02.2018   wieslaw.luczynski@gmail.com
www.sites.google.com/site/fencingsport/

senjabl, 2018-03-02:
I need some time to make documentation, drawings, files, etc...

bhupilm, 2018-03-13:
Can you send me video how to make

senjabl, 2018-03-13:
@bhupilm I still have some documentation to finish, like shematic, PCB, etc. I will publish it in my fork on github. There is already temporal code for timer part of scoring box. Right now I am waiting for some components, since IR remote has pathetic range, so i will switch to bluetooth.

Solexis2001, 2018-03-14:
@senjabl, did you make the sabre part work ? Did you add condensators to make it work ?

senjabl, 2018-03-14:
Sorry. It is tested only with epee and partly with floret. We don't fence with sabre. To be honest I held it in my hand only once in my life.

senjabl, 2018-03-14:
My changes are on:

https://github.com/senjabl/fencing_scoring_box

bhupilm, 2018-03-15:
Sir send me a video how make it
Regard
Bhupinder Singh

senjabl, 2018-03-15:
@bhupilm I will not make video. I really don't have time for this and besides that, I will have to completely dismantle whole unit and it will not happen. I will publish written instructions with some photos.

bhupilm, 2018-03-15:
Ok dear thank you you have software if so please send me
Regard
Bhupinder Singh

senjabl, 2018-03-15:
@bhupilm  Again, firmwares for both arduinos are on https://github.com/senjabl/fencing_scoring_box

bhupilm, 2018-03-15:
Ok thank you
Regard
Bhupinder Singh