wnew/fencing_scoring_box #13: Hi, can help me read the Schematic?

gravataLonga, 2018-08-30:
Hello, i'm noob reading Schematic, can explain me more about that? Or make a video o a picture of Schematic mounted? 

I'm a full stack developer, but this beyond my expertise, any help it will appreciate!
Thanks a lot for this project!

chronossage, 2018-09-26:
the schematic on here is very basic.  it shows the arduino pens going to the corresponding led or input.

seigel, 2018-09-26:
@gravataLonga Sir, if you send me a link to the schematic in question I can do my best to explain it to you in terms that might work for you.  I am a full stack dev as well, so I might speak your kind of geek! :)

chronossage, 2018-09-26:
Here is a copy of a schematic that I created for this project.

seigel, 2018-09-26:
You might have to add a link to the image here somewhere.

chronossage, 2018-09-26:
I dont have s specific image for the box because i custom created the
pcb for them.

Irmarek, 2019-05-07:
First of all thanks for developing this topic. I am a fan of fencing, especially epee. I try to run the epee fencing scoring apparatus made according to description but unfortunately it does not blocked hits in the guard or piste. Hence I have a doubt:  since both analog inputs A0 and A5 on are connected to the ground by a 1K resistor, so how will possible that 1.7V voltage appear here during hit on guard, or how will be possible detect hit in grounded a piste? At all times A0 and A5 are conected to ground - so how? Is it correct? I am a beginner - I will be grateful for short explanation.

closed 2024-07-18 by wnew