wnew/fencing_scoring_box #16: code logic may have a problem

guyrogel, 2020-05-05:
After using the scoring machine today (Fencing Foil)  I noticed 2 problems :
1. There is a delay between point pressed and the light coming up 
2. when 2 hits are done in sequence within the lockout time both light come at the same time (although the hits did not came a the exact same time.

This made me look into the code again and I have realized the logic is not correct.
The code starts the signal hits only after the lockout time is finished.
But is should show the light immediately and not wait.
This delay is wrong and can prevent the referrer from taking the correct decisions
I will try to fix the code to work better and upload it here
Guy

guyrogel, 2020-05-07:
Hi all
I attach an updated code we have made
the main difference is the fact the signals turn on immediately not waiting for the lockout time to finish. I believe this is much more correct. this is how the professional machines work.
Also we fixed a bug in the mode selecting part
Tested on foil - works perfect
Guy
[fencing_code_mod.zip](https://github.com/wnew/fencing_scoring_box/files/4595487/fencing_code_mod.zip)

nova0205, 2024-06-14:
hi 
the only thing that you've changed is the position of the call of the function signalhits(), isnt it?

wnew, 2024-06-18:
Hi Guys.

@guyrogel Sorry for not picking this up earlier. You are correct that there is a slight delay in turning the lights on as the code waits for the lockout to occur. Ill take a look at this and will sort it out. Unfortunately the code you shared has other changed in it that are not relevant. Ill see how I can merge them.

I am also unsure how a ref will look at which light came on first when he is concentrating on the fencers and "right of way" is the determining factor, not which light came on first. Can you explain further?

@nova0205 it is not only the signalhits, that was just changed so the lights are correct on startup.

guyrogel, 2024-06-20:
Hi Nova
In foil fencing the refery must look both on the machine and the fenders, surly the sound helps. Both fencers and the refery must get an imidate respond when touching.
As it has been a long time, I can not remember what other changes we have done. But the code i published works well for us.

wnew, 2024-06-20:
Yes, it needs to be sorted out. When I have the time to test it Ill make the changes to the code. Thanks for sharing your changes.

wnew, 2024-06-20:
@guyrogel I havent got any hardware available to me at the moment. Would you be able to test some changes for me?

nova0205, 2024-06-28:
i have the hardware if u want to test it

wnew, 2024-07-18:
Thanks @nova0205 I have put together a new PCB and tested it that way. I have updated the code. It is all in the devel branch for now. 

Here is a taste of the new PCB:
[image]

wnew, 2024-07-18:
This is fixed in the devel branch and will be merged to master in the coming weeks.

closed 2024-07-18 by wnew